octo-models/octo #10: No module named 'dlimp'

faran332, 2023-12-15:
Hello, 

I have installed octo as instructed. When I wanted to test the dataloading I got the error: 
`ModuleNotFoundError:` No module named 'dlimp'`
Inside the requirements.txt there is this line:
`dlimp @ git+https://github.com/kvablack/dlimp@d08da3852c149548aaa8551186d619d87375df08`
When i followed that link it leads to a non existing repo.

faran332, 2023-12-15:
I managed to fix it using 

pip install - e git+https://github.com/kvablack/dlimp@d08da3852c149548aaa8551186d619d87375df08#egg=dlimp

closed 2023-12-15 by faran332

kpertsch, 2023-12-15:
Glad you were able to fix it! It should also get installed automatically if you run `pip install -r requirements.txt`
Here is the dlimp repo for reference: [https://github.com/kvablack/dlimp](https://github.com/kvablack/dlimp)