octo-models/octo #11: Can't get primary + wrist camera

seann999, 2023-12-16:
I'm trying to finetune Octo on custom data by modifying a script in the examples:

```
dataset = make_single_dataset(
    dataset_kwargs=dict(
        name="custom",
        data_dir=FLAGS.data_dir,
        image_obs_keys={"primary": "head", "wrist": "hand"},
        state_obs_keys=["state"],
        language_key="language_instruction",
        action_proprio_normalization_type=NormalizationType.NORMAL,
        absolute_action_mask=[True] * 11,
    ),
    traj_transform_kwargs=dict(
        window_size=1,
        future_action_window_size=49,  # so we get 50 actions for our action chunk
    ),
    frame_transform_kwargs=dict(
        resize_size={"primary": (256, 256), "wrist": (128, 128)},
    ),
    train=True,
)
```
The original data has head and hand images both at 224x224, 6-D proprioception, and an 11-D action space. But the output seems to indicate that the wrist image is being ignored. What might I be doing incorrectly?

```
┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃                                ┃ task_language (16 tokens) ┃ t=0 obs_primary (256 tokens)  ┃ t=0 obs_proprio (6 tokens)  ┃ t=0 readout_action (1 tokens)  ┃
┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│ task_language (16 tokens)      │ x                         │ x                             │ x                           │ x                              │
├────────────────────────────────┼───────────────────────────┼───────────────────────────────┼─────────────────────────────┼────────────────────────────────┤
│ t=0 obs_primary (256 tokens)   │                           │ x                             │ x                           │ x                              │
├────────────────────────────────┼───────────────────────────┼───────────────────────────────┼─────────────────────────────┼────────────────────────────────┤
│ t=0 obs_proprio (6 tokens)     │                           │ x                             │ x                           │ x                              │
├────────────────────────────────┼───────────────────────────┼───────────────────────────────┼─────────────────────────────┼────────────────────────────────┤
│ t=0 readout_action (1 tokens)  │                           │                               │                             │ x                              │
└────────────────────────────────┴───────────────────────────┴───────────────────────────────┴─────────────────────────────┴────────────────────────────────┘
```

kvablack, 2023-12-16:
Hi @seann999,

Your dataset config looks good --- it should be pulling out both the "head" image (mapped to "image_primary") as well as the "hand" image (mapped to "image_wrist"). However, the second thing you pasted is a printout from the model side, which shows what tokenizers and readouts the model itself is configured with.

I assume you're building off of the example [02_finetune_new_observation_action.py](https://github.com/octo-models/octo/blob/main/examples/02_finetune_new_observation_action.py)? That example deliberately overrides the model config to remove the wrist camera, adding a proprioceptive input instead:

```python
    # load pre-training config and modify --> remove wrist cam, add proprio input, change action head
    # following Zhao et al. we use "action chunks" of length 50 and L1 loss for ALOHA
    config = pretrained_model.config
    del config["model"]["observation_tokenizers"]["wrist"]
    ###
    config["model"]["observation_tokenizers"]["proprio"] = ModuleSpec.create(
        LowdimObsTokenizer,
        n_bins=256,
        bin_type="normal",
        low=-2.0,
        high=2.0,
        obs_keys=["proprio"],
    )
```

If you remove the `del config["model"]["observation_tokenizers"]["wrist"]` line, your finetuning script should continue to process the wrist camera input.

seann999, 2023-12-16:
Yes, I was building off of [02_finetune_new_observation_action.py](https://github.com/octo-models/octo/blob/main/examples/02_finetune_new_observation_action.py).
I had totally missed that line; removing it as you said solved the problem.
Thanks!

closed 2023-12-16 by seann999

Malak-Mansour, 2025-04-21:
Hi @seann999, I see that you finetuned octo on your custom dataset. I am not sure what data type you are using but I am trying to finetune octo on the libero_object dataset, which is in the .hdf5 format, while octo is expecting rlds. 
Also @kvablack if you have any ideas, I would appreciate any guidance on how I can achieve this task, whether it be converting .hdf5 to rlds, or if the current finetuning code can be adapted to work with .hdf5.