octo-models/octo #13: Full fine-tuning in 02_finetune example

seann999, 2023-12-19:
Hi, I'm trying to train all parameters by modifying the example, but it seems like it's not working for me. What might be the issue?

With the original script, I get the following log:

```
 298 I1218 11:34:09.575279 140267533906560 train_utils.py:403] ########## Parameters skipped during model loading: ##########
 299 I1218 11:34:09.575732 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.LayerNorm_0.bias
 300 I1218 11:34:09.575860 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.LayerNorm_0.scale
 301 I1218 11:34:09.575954 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MlpBlock_0.Dense_0.bias
 302 I1218 11:34:09.576039 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MlpBlock_0.Dense_0.kernel
 303 I1218 11:34:09.576118 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MlpBlock_0.Dense_1.bias
 304 I1218 11:34:09.576195 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MlpBlock_0.Dense_1.kernel
 305 I1218 11:34:09.576265 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.key.bias
 306 I1218 11:34:09.576333 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.key.kernel
 307 I1218 11:34:09.576400 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.out.bias
 308 I1218 11:34:09.576465 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.out.kernel
 309 I1218 11:34:09.576580 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.query.bias
 310 I1218 11:34:09.576655 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.query.kernel
 311 I1218 11:34:09.576721 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.value.bias
 312 I1218 11:34:09.576784 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.value.kernel
 313 I1218 11:34:09.576846 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.probe
 314 I1218 11:34:09.576910 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.mean_proj.bias
 315 I1218 11:34:09.576974 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.mean_proj.kernel
 316 I1218 11:34:09.577038 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: octo_transformer.obs_proprio_pos_embedding
 317 I1218 11:34:09.577101 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: octo_transformer.obs_proprio_projection.bias
 318 I1218 11:34:09.577165 140267533906560 train_utils.py:405] Param missing in pre-trained model, skipping: octo_transformer.obs_proprio_projection.kernel
 319 I1218 11:34:09.578479 140267533906560 train_utils.py:248] Freezing parameters that include the following keys: ['*hf_model*'].
 320 I1218 11:34:09.581623 140267533906560 train_utils.py:284] Num trainable params: 27,178,734.
 321 I1218 11:34:09.581736 140267533906560 train_utils.py:285] Num frozen params: 109,628,544.
 322 I1218 11:34:09.581822 140267533906560 train_utils.py:286] To see a detailed list of frozen params, set logging level to DEBUG.
 323 I1218 11:34:10.776679 140267533906560 02_finetune_new_observation_action.py:186] Starting finetuning...
```

Then, I set [frozen_keys](https://github.com/octo-models/octo/blob/37951e4e6d708fd76374f6e09e716763fe2673b1/examples/02_finetune_new_observation_action.py#L149C4-L149C4) to an empty list in a crude attempt to fine-tune fully. I got the following with this:

```
 305 I1218 18:19:57.928706 139778262782592 train_utils.py:403] ########## Parameters skipped during model loading: ##########
 306 I1218 18:19:57.929069 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.LayerNorm_0.bias
 307 I1218 18:19:57.929162 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.LayerNorm_0.scale
 308 I1218 18:19:57.929238 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MlpBlock_0.Dense_0.bias
 309 I1218 18:19:57.929309 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MlpBlock_0.Dense_0.kernel
 310 I1218 18:19:57.929384 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MlpBlock_0.Dense_1.bias
 311 I1218 18:19:57.929462 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MlpBlock_0.Dense_1.kernel
 312 I1218 18:19:57.929538 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.key.bias
 313 I1218 18:19:57.929613 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.key.kernel
 314 I1218 18:19:57.929690 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.out.bias
 315 I1218 18:19:57.929764 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.out.kernel
 316 I1218 18:19:57.929843 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.query.bias
 317 I1218 18:19:57.929908 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.query.kernel
 318 I1218 18:19:57.929970 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.value.bias
 319 I1218 18:19:57.930034 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.MultiHeadDotProductAttention_0.value.kernel
 320 I1218 18:19:57.930099 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.map_head.probe
 321 I1218 18:19:57.930162 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.mean_proj.bias
 322 I1218 18:19:57.930227 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: heads_action.mean_proj.kernel
 323 I1218 18:19:57.930290 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: octo_transformer.obs_proprio_pos_embedding
 324 I1218 18:19:57.930354 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: octo_transformer.obs_proprio_projection.bias
 325 I1218 18:19:57.930418 139778262782592 train_utils.py:405] Param missing in pre-trained model, skipping: octo_transformer.obs_proprio_projection.kernel
 326 I1218 18:19:57.931729 139778262782592 train_utils.py:248] Freezing parameters that include the following keys: [].
 327 I1218 18:19:57.934117 139778262782592 train_utils.py:284] Num trainable params: 136,807,278.
 328 I1218 18:19:57.934224 139778262782592 train_utils.py:285] Num frozen params: 0.
 329 I1218 18:19:57.934307 139778262782592 train_utils.py:286] To see a detailed list of frozen params, set logging level to DEBUG.
 330 I1218 18:19:59.401661 139778262782592 02_finetune_new_observation_action.py:186] Starting finetuning...
```

0 frozen params, so this seemed right. However, VRAM usage did not change (honestly I thought I would get GPU OOM with this), and the loss curve remained mostly unchanged from the default script:

[image]

So I'm guessing I'm missing/misunderstanding something?

dibyaghosh, 2023-12-19:
Great question! Right now, out of technical convenience, we're bundling the T5 language encoder parameters with the rest of the network (this is what the 109M frozen parameters are). In our code / report, when we say "full fine-tuning", what we really mean is fine-tuning everything but the T5 language encoder. So, your first script is already fine-tuning everything (the observation tokenizers, the transformer, and the action head).

If you're doing a language-conditioned task, and you'd actually like to fine-tune the T5 language encoder (we didn't find this helped too much ourselves), then you can do so by doing the following two things together:

1. As you've already done, manually set `frozen_keys` to an empty list
2. Change the configuration of the task tokenizer to set `finetune_encoder = True` here: https://github.com/octo-models/octo/blob/37951e4e6d708fd76374f6e09e716763fe2673b1/octo/model/components/tokenizers.py#L181

seann999, 2023-12-19:
Ah, that makes sense. Yeah I just checked and realized 27M is the parameter count of the policy. So I was already doing what I wanted to do. Thanks!

closed 2023-12-19 by seann999