octo-models/octo #20: Discuss the zero-shot capability of the model in the paper

swearos, 2023-12-25:
The zero-shot capability of the model is mentioned several times in the paper and compared with the RT-1-X and RT-2-X respectively

1. In the evaluation experiment, how to define the zero-shot capability?
(The operation of objects is different , but  robotic arm in the evaluation stage has appeared in the training data?)

2. In the comparison data of the experiment, is RT-1-X using the open source model, and RT-2-X is the data of the cited paper?

Thank you!

kpertsch, 2023-12-25:
1. Yes, we refer to experiments as "zero-shot" if we don't perform any finetuning after Open-X pre-training. This assumes that the robot setup used for eval was seen during Open-X pre-training so the model is familiar with the used action space definition. We do test generalization to new objects etc in these experiments though as you mentioned.

2. Yes, we are using the official RT-1-X checkpoint and take the numbers for RT-2-X from the SuSIE paper (Black et al.) and evaluate other methods on the same tasks used in their evals.

swearos, 2023-12-27:
Some questions related to RT-1-X:

You provided inference demo similar to [RT-1-X colab scripts](https://colab.research.google.com/github/google-deepmind/open_x_embodiment/blob/main/colabs/Minimal_example_for_running_inference_using_RT_1_X_TF_using_tensorflow_datasets.ipynb#scrollTo=6Wwo5cdhT23A), all running on the bridge dataset. 

RT-1-x compared with your version, the biggest difference is that **language_instruction is not added in the inference, is there a problem here?**

I tried to add natural_language_instruction and natural_language_embedding to the observation, but found that the action had not changed.

so can you give the code snippeting your call to RT-1-X,thanks !

mees, 2023-12-29:
Hi there,
thanks for trying out Octo!
I am not sure if I understand your question correctly, but in our [inference colab](https://colab.research.google.com/drive/1z0vELj_lX9OWeoMG_WvXnQs43aPOEAhz?usp=sharing) we showcase how to specify goals with goal images or language instructions:
```
task = model.create_tasks(goals={"image_primary": goal_image[None]})   # for goal-conditioned
task = model.create_tasks(texts=[language_instruction])                # for language conditioned
```
thus the default conditioning in the colab is currently language, as it overrides the first task definition. If you comment out the second line and re-run the inference with the goal image, you will see that naturally the predicted actions vary slightly w.r.t to language conditioning. Hope that solves your issue!

swearos, 2023-12-29:
@mees  
Thank you for your reply, I have no questions about your OCTO inference code, your work is very helpful to me and the community.

I only have a few questions about the experimental data mentioned in your paper for model runs using RT-1-X:

What I mean is: **Is there a problem with the inference script of [RT-1-x](https://colab.research.google.com/github/google-deepmind/open_x_embodiment/blob/main/colabs/Minimal_example_for_running_inference_using_RT_1_X_TF_using_tensorflow_datasets.ipynb#scrollTo=6Wwo5cdhT23A):** The same inference on the bridge dataset, without adding instructions, RT-1-X can get action; And after I added the instruction, the action did not change the task.
 
Is there a problem here? Did you pay attention to this problem when you used rt-1-x?

mees, 2023-12-29:
The RT-1 colab you link seems to be doing inference with an empty language embedding, which is probably a small  oversight by the authors. We made sure to condition the RT models properly in our experiments. If you have questions regarding the RT models, reaching out to the authors might be best!
Hope this answers your questions regarding Octo!

Closing this issue for now.

closed 2023-12-29 by mees