octo-models/octo #24: Run on new robot ?

apirrone, 2023-12-29:
Hi, 

Great work ! And thank you very much for sharing the code and pretrained weights !

After browsing the repo and project page, I am a little bit confused on how I could apply octo to a new robot. 

I would like to run it on a humanoid robot (Reachy https://www.pollen-robotics.com/), which has a geometry that is quite different from the robot arms you showcase. 

I guess I would need to collect data for fine tuning and provide a description of the robot to the system somehow, as well as write some kind of interface ? But it is not clear to me what are the expected formats for the data, the robot description (URDF?) etc. 

Could you give an overview of the required steps ?

Thank you vey much !

mees, 2023-12-29:
Thanks for your interest and the great question!

1. Yes, you need to record some expert demonstrations and store them in the [RLDS format](https://github.com/google-research/rlds). If you have an existing dataset, you can check [this example](https://github.com/kpertsch/bridge_rlds_builder/) on how to do the conversion.
2. Once you have a target dataset, you can start by trying finetuning to your robot, by specifying your observation and action space. A great example on how to do this is [this notebook](https://github.com/octo-models/octo/blob/main/examples/02_finetune_new_observation_action.py) where we show how to finetune to a bimanual setup with joint control.

Does this answer your question?

apirrone, 2023-12-29:
Yes thanks ! 

We probably try it out in the following months, what amount of expert demonstration data would you consider sufficient for fine tuning on a new robot ?

mees, 2023-12-29:
Hmm that's very dependant on the task, the action and observation space etc.  The finetuning experiments in the technical report used around 100 in-domain demos.

closed 2023-12-29 by mees