octo-models/octo #27: Random / Noisy behavior of a new robot

HM102, 2024-01-03:
Hello,

```
Setup: new environment
Robot: Franka panda
Data de-normalization:  I am using the Berkeley cable routing as it also uses Franka robot ( would you recommend different one?)
```

I am trying to run your base model on a Franka panda robot. I have not done fine tuning yet. I wanted to check and test the initial behavior first. 

For now I am getting random actions from the robot and some times fixed noisy actions( always moving down, no matter what the instruction/goal image is or even the robot initial position). It is not succeeding for simple tasks like move to object or pick up object.

Is this the expected behavior? or it should at least move in the direction of the object?

Thank you .

kpertsch, 2024-01-19:
Sorry for the late reply!
At the moment the Octo model is not conditioned on any action space information so the model is not expected to work 0-shot for setups that are not present in the training data. Since your setup (e.g. control stack, action space definition etc) is not present in our pre-training mix you'd need to collect a small finetuning dataset in your target setup and that should hopefully give you more reasonable behavior.

closed 2024-01-30 by HM102

victorbjo, 2024-03-07:
@HM102 How did you go about getting the Franka working with Octo? Would you be willing to share your code?