octo-models/octo #33: MSE loss in logs

lyshuga, 2024-01-11:
Hi,

After doing fine-tuning, I have checked the values in two tabs: "
offline_metrics_red_cube/text_conditioned" and "training", and they were quite different.

[image]

I was wondering if it is the same function used in both of evaluations?

kpertsch, 2024-01-11:
The training "MSE" for the diffusion head we are using is actually the MSE between predicted and actual *noise*, not actions. In contrast, the evaluation MSE is the mse between predicted and actual *actions*, so they are not comparable. See training MSE computation [here](https://github.com/octo-models/octo/blob/8fe74973c37516a2ee7585544bb5d4e7b7da059e/octo/model/components/action_heads.py#L555C49-L555C49).

This is a bit confusing, maybe we should proper action MSE computation to training or remove the training MSE for diffusion head alltogether.

lyshuga, 2024-01-11:
Thanks for your reply, clear now.

Indeed, it would be great at least to add action MSE to the train logs, and split up the names. Should be easy to understand after that :)