octo-models/octo #35: Hugging face description for octo-small may be incorrect

PeterMitrano, 2024-01-15:
From inspecting the model, it seems there actually is no "wrist" observation tokenizer for the small model, only the "base" model. Also, I'm only seeing the "language" task tokenizer for both the "base" and "small" models. I'm checking by looking in `pretarined_model.config['model'`.

Perhaps the descriptions on HF should be updated?

kpertsch, 2024-01-19:
I just tried the following and the config seems to indicate correctly that the Octo small model has a wrist encoder?
For task tokenization: the goal images are channel stacked in the Image tokenizers (see `task_stack_keys`), so they don't have a separate task tokenizer.

```
from octo.model.octo_model import OctoModel
model = OctoModel.load_pretrained("hf://rail-berkeley/octo-small")
print(json.dumps(model.config['model'], indent=4))

{
    "observation_tokenizers": {
        "primary": {
            "module": "octo.model.components.tokenizers",
            "name": "ImageTokenizer",
            "args": [],
            "kwargs": {
                "obs_stack_keys": [
                    "image_primary"
                ],
                "task_stack_keys": [
                    "image_primary"
                ],
                "encoder": {
                    "module": "octo.model.components.vit_encoders",
                    "name": "SmallStem16",
                    "args": [],
                    "kwargs": {}
                }
            }
        },
        "wrist": {
            "module": "octo.model.components.tokenizers",
            "name": "ImageTokenizer",
            "args": [],
            "kwargs": {
                "obs_stack_keys": [
                    "image_wrist"
                ],
                "task_stack_keys": [
                    "image_wrist"
                ],
                "encoder": {
                    "module": "octo.model.components.vit_encoders",
                    "name": "SmallStem16",
                    "args": [],
                    "kwargs": {}
                }
            }
        }
    },
    "task_tokenizers": {
        "language": {
            "module": "octo.model.components.tokenizers",
            "name": "LanguageTokenizer",
            "args": [],
            "kwargs": {
                "encoder": "t5-base",
                "finetune_encoder": false
            }
        }
    },
    "heads": {
        "action": {
            "module": "octo.model.components.action_heads",
            "name": "DiffusionActionHead",
            "args": [],
            "kwargs": {
                "readout_key": "readout_action",
                "use_map": false,
                "pred_horizon": 4,
                "action_dim": 7
            }
        }
    },
    "readouts": {
        "action": 1
    },
    "token_embedding_size": 384,
    "transformer_kwargs": {
        "attention_dropout_rate": 0.0,
        "add_position_embedding": false,
        "num_layers": 12,
        "mlp_dim": 1536,
        "num_attention_heads": 6,
        "dropout_rate": 0.0
    },
    "max_horizon": 10
}
```

PeterMitrano, 2024-01-20:
You're right, I don't know what I saw before but it must have been a mistake. Thanks!

closed 2024-01-20 by PeterMitrano