octo-models/octo #43: Issue with diffusion head

seann999, 2024-01-29:
I was able to fine-tune with a modified version of example 2 with the following action head:
```
config["model"]["heads"]["action"] = ModuleSpec.create(
    L1ActionHead,
    pred_horizon=9,
    action_dim=11,
    readout_key="readout_action",
)
```

The policy works reasonably well on the robot.
After, I've been trying to fine-tune with a diffusion head, but the robot goes out of control with this.

```
config["model"]["heads"]["action"] = ModuleSpec.create(
    # L1ActionHead,
    DiffusionActionHead,
    use_map=False,

    pred_horizon=9,
    action_dim=11,
    readout_key="readout_action",
)
```

The rest of the script is largely unchanged besides data augmentation, which I am currently controlling to debug this issue.
What else could be the problem?

seann999, 2024-01-29:
Okay, I was able to figure out that the horizon was too large; 1~3 was fine but 9 led to an unstable model.

seann999, 2024-01-29:
Reopening since although the output values are more stable, I found that the policy is still considerably worse than an L1-based policy.

kpertsch, 2024-01-29:
Thanks for digging into this!
We are debugging the ALOHA finetuning on our side as well, and also found that the diffusion head is somehow substantially worse than the L1 head in this particular case and outputs non-sensical values. I think it should work *in principle* so we are digging into this as well. Will update here if we find a solution!

seann999, 2024-01-31:
With a diffusion head and pred_horizon=4, I was able to get a sufficiently good policy by:
* changing the optimizer and its hyperparameters from those of [02.py](https://github.com/octo-models/octo/blob/653c54acde686fde619855f2eac0dd6edad7116b/examples/02_finetune_new_observation_action.py#L145-L147) to those of [finetune_config.py](https://github.com/octo-models/octo/blob/653c54acde686fde619855f2eac0dd6edad7116b/scripts/configs/finetune_config.py#L70C9-L83C11) (larger peak LR, cosine schedule, gradient clipping).
* training longer; changed from 50k steps to 300k steps

However, with the same config but with pred_horizon=10, I still can't get a good policy. The values aren't extreme anymore, as the robot moves smoothly instead of thrashing. But it fails at fairly simple tasks like grasping a ball, which can be successfully done by the policy with pred_horizon=4.

zwbx, 2024-04-08:
Similar problem. The paper indicate that the Diffusion head outperforms the L1 head. I guess, this conclusion is not necessarily extended to the fintuing condition.

zwbx, 2024-04-08:
Maybe we can build a Discord to discuss these questions, the official reply is too slow these days.

BUAAZhangHaonan, 2024-05-11:
> Thanks for digging into this! We are debugging the ALOHA finetuning on our side as well, and also found that the diffusion head is somehow substantially worse than the L1 head in this particular [...]

Compared with the L1 and MSE heads, the diffusion head seems to show more unstable characteristics under fewer training steps. In MuJoCO, simulation instability will occur, resulting in the following error:
```
Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. Time = 0.0840.
```
```
dm_control.rl.control.PhysicsError: Physics state is invalid. Warning(s) raised: mjWARN_BADQACC
```
Even reducing the size of the horzion parameter still has similar problems, and it seems that using a diffusion head for fine-tuning is more expensive. I don't know whether the head parameters in the config should be adjusted more carefully, or whether this problem is simply a flaw in the diffusion algorithm.

zwbx, 2024-05-13:
Hi, 

> Compared with the L1 and MSE heads, the diffusion head seems to show more unstable characteristics under fewer training steps. In MuJoCO, simulation instability will occur, resulting in the [...]

Hi, have you checked the reason of the error? Try to print the action the model output, you could see that the number is out of bound or hard to be solved by IK optimizer.

BUAAZhangHaonan, 2024-05-13:
> Hi,
>
>
> Hi, have you checked the reason of the error? Try to print the action the model output, you could see that the number is out of bound or hard to be solved by IK optimizer.

Thank you for your attention. I tried Aloha simulation with 1000 and 5000 steps on Diffusion. Unfortunately, this problem was not reproduced. Maybe this problem itself is also unstable like Diffusion Head.
But I do get action prediction output from two different heads:
**The first is the output of L1, which can be simulated normally and performs well:**
action[100]: [ 0.57204974 -0.23319605 0.76363015 -0.09170198 -0.07699968 0.77580905 0.96044242 0.75630909 -0.45451248 1.13872492 -0.46 336147 -0.58958995 0.5670839 1.06122994]
action[200]: [-0.99930573 0.55937028 -0.52284944 -0.59289199 -0.51960701 0.75874555 0.97301495 0.14920622 0.19651346 -0.29161143 -0.02 436048 0.51398659 0.05542321 -1.52965879]
  75%|action[300]: [-0.46441016 0.91491914 -1.4249717 -0.39001578 -0.50952721 0.77543199 -0.09499384 -0.16243899 0.69015974 -1.14362192 0.15943812 0.9602797 -0.15596175 0.2628575 ]
**Then there is the output of Diffusion, which completely freezes after 100 steps and maintains consistent movement:**
action[100]: [-3.59837627 5. 5. -4.92470407 5. 4.25408506 -4.62254953 -4.55121994 -5. -5. 1.16204941 4.00920582 -4.54648066 5. ]
action[200]: [-3.59454226 5. 5. -4.92671156 5. 4.25115156 -4.61913872 -4.55262327 -5. -5. 1.16107273 4.01593781 -4.55090284 5. ]
action[300]: [-3.59459043 5. 5. -4.92677879 5. 4.25059795 -4.61849213 -4.55246067 -5. -5. 1.16351843 4.01590443 -4.55066919 5. ]
Compared with the two, the latter does have some extreme values, showing its instability.

nakamotoo, 2024-10-30:
> Thanks for digging into this! We are debugging the ALOHA finetuning on our side as well, and also found that the diffusion head is somehow substantially worse than the L1 head in this particular [...]

For ppl who are experiencing this issue: try using the newly added UNet version of the DDPM action head
https://github.com/octo-models/octo/blob/241fb3514b7c40957a86d869fecb7c7fc353f540/octo/model/components/action_heads.py#L613

zwbx, 2025-06-07:
Hello everyone,

I recently published a paper on Generalist Robot Manipulation Policies (GMPs) at ICRA 2025, focusing on the impact of fine-tuning strategies for Octo. You can find it [https://arxiv.org/abs/2410.01220](https://arxiv.org/abs/2410.01220).

In our experiments, we found that the Diffusion Head struggles with fine-tuning. It often produces unstable action values, leading to poor performance compared to the L1 Head. This highlights the need for careful strategy selection when fine-tuning GMPs.

Our study provides practical guidelines for finetune Octo. I hope this research helps others working on similar challenges. Feel free to reach out if you have any questions!

JuilieZ, 2025-06-07:
> Hello everyone,
>
> I recently published a paper on Generalist Robot Manipulation Policies (GMPs) at ICRA 2025, focusing on the impact of fine-tuning strategies for Octo. You can find it [...]

Hi wb,thx for your wonderful work and congrats! In your paper, is the fine-tuning performed **separately for each individual task**, or is it done **jointly for all 10 tasks at once**?

zwbx, 2025-06-17:
Hi, thanks for your attention. All experiments are conducted in single-task training mode.