octo-models/octo #44: Bug in the finetune_new_observation_action.py

HM102, 2024-01-30:
Whether I set this to `False` or `True` I am getting the same number of parameters:

```
Flags.DEFINE_bool(
    "freeze_transformer",
    True,
    "Whether pre-trained transformer weights should be frozen.",
)
```

When it is set to` False`, I get  in the log:

```
Freezing parameters that include the following keys: ['*hf_model*'].
Num trainable params: 25,639,496.
Num frozen params: 109,628,544.
```

If I set it to `True` :
```
Freezing parameters that include the following keys: ['*hf_model*', 'head_mlp_only'].
Num trainable params: 25,639,496.
Num frozen params: 109,628,544.
```

Both have same number of parameters!

kpertsch, 2024-01-30:
Ah this is a good catch!
Could you try setting [this](https://github.com/octo-models/octo/blob/653c54acde686fde619855f2eac0dd6edad7116b/examples/02_finetune_new_observation_action.py#L151) line to 
```
frozen_keys.append("octo_transformer.BlockTransformer_0.*",)
```
and try again?

HM102, 2024-02-07:
Yes this solves it. Thank you

closed 2024-02-07 by HM102