octo-models/octo #47: Stl for the wrist camera holder

HM102, 2024-02-07:
Can you provide the stl to 3D print the wrist camera holder for the franka panda robot.

kpertsch, 2024-02-07:
The FMB benchmark should have STL files for the RealSense camera mount used in the Franka peg insertion experiments: https://functional-manipulation-benchmark.github.io/

Does that work?

HM102, 2024-02-08:
Thank you for the replay. I should have more specific. 

This holder you sent is for realsense d405 which is a square one. I need one for the d455 or d435.

In addition, while on the topic of realsense, can you input to octo 3 images? Wrist, 3rd person and depth.  If not, which is better to use as a secondary image depth or wrist?

lucanunz, 2024-10-24:
I have the same request as @HM102. Any updates on this?