octo-models/octo #50: Third-person wrist image in the paper

HM102, 2024-02-19:
In the paper, in the last page (`19)` it is mentioned that in the coffee demo a `single third-person wrist` were used. What do you mean by this? Does this contradict with what you said in the paper regarding wrist images achieving worse performance due to lack of data.

jhejna, 2024-02-19:
sorry about this -- its a typo. It should say "a single third-person observation". The pod picking and placement was all done within a single episode -- ie we collected demos of picking and placing and didn't treat them as separate tasks.

HM102, 2024-02-19:
Thank you.

Csn you share the position of the third person camera and the language prompt used. Since I am having some difficulty getting the fine tuned working on this demo.

jhejna, 2024-02-19:
1. We used the goal conditioned version of the policy for this task (and all other finetuning tasks)
2. Resolution is very important for this since the motion is precise. 
3. Demonstration quality is also very important. to get this task to work I originally collected around 200 demos and ended up manually filtering them down to only contain high quality ones.

Here is an example video of a robot trial recorded from the observation camera.

https://github.com/octo-models/octo/assets/28748211/aec6ba99-1d29-4497-804f-74ebcb2aac28

HM102, 2024-02-19:
Thank you for the response. 

The video link is broken.

jhejna, 2024-02-19:
Fixed the video!

Quality was the following:
1. Consistently grabbing the pod in the same way (ie similar depth)
2. Consistently placing using the same motion (reducing some multimodality). Some directions of the robot movement occluded things more.
3. placing precisely, and including some adjustments in the set of demonstrations
4. Making sure the pod and coffee maker was fully in view at all times, since the policy does not take in other inputs like proprioception etc.

HM102, 2024-02-19:
Thank you very much for detailed feedback! Well take them into consideration!

closed 2024-02-19 by HM102