octo-models/octo #56: Google Colab not working

victorbjo, 2024-02-28:
Google Colab from link: https://colab.research.google.com/drive/1z0vELj_lX9OWeoMG_WvXnQs43aPOEAhz?usp=sharing
can not run. This line in step 1 `action = model.sample_actions(observation, task, rng=jax.random.PRNGKey(0))` seems to be the issue. I get this error
```---------------------------------------------------------------------------

AssertionError                            Traceback (most recent call last)

[<ipython-input-23-97a79f13d085>](https://localhost:8080/#) in <cell line: 8>()
      6 task = model.create_tasks(texts=["pick up the fork"])
      7 
----> 8 action = model.sample_actions(observation, task, rng=3)#jax.random.PRNGKey(0))
      9 print(action)   # [batch, action_chunk, action_dim]

    [... skipping hidden 12 frame]

2 frames

    [... skipping hidden 12 frame]

[/content/octo/octo/model/octo_model.py](https://localhost:8080/#) in _verify_shapes(pytree, name, example_pytree, starting_dim, strict, raise_error, silent)
    488 
    489     if raise_error and (fail or (weak_fail and strict)):
--> 490         raise AssertionError(f"{name} does not match example batch.")
    491 
    492     return weak_fail or fail

AssertionError: observations does not match example batch.```

kpertsch, 2024-02-29:
I just ran the colab from our website "as-is" and it ran without any problem -- did you maybe make modifications to it? can you try running a freshly pulled colab from our website?

[image]

kpertsch, 2024-02-29:
(note that I was also running on a T4)

victorbjo, 2024-03-06:
I did not get it working, even when trying again from scratch. I had a friend of mine try and it worked for him, so I guess the problem is all mine :D

closed 2024-03-06 by victorbjo