octo-models/octo #65: Installation and compilation issues

hahamini, 2024-03-18:
- Env
  - RTX 4090
  - Ubuntu20.04
  - Python -> Used virtual env  (python3.10) 
  - CUDA & cudnn in local: cuda-11.2, cudnn 8.6.0

I installed octo in my virtualenv using pip as instructed in Readme.  
After that, when I run the test code, I can't proceed any further with the message "Very slow compile?"  
I also tried reinstalling jax, but I don't know the cause.
How can I solve this?

```text
octo/scripts/finetune.py:3: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
  import imp
2024-03-18 10:58:02.165197: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2024-03-18 10:58:02.165237: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2024-03-18 10:58:02.166325: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2024-03-18 10:58:02.959024: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
2024-03-18 10:58:05.042673: E external/local_xla/xla/stream_executor/cuda/cuda_driver.cc:274] failed call to cuInit: CUDA_ERROR_SYSTEM_DRIVER_MISMATCH: system has unsupported display driver / cuda driver combination
W0318 10:58:05.043408 140289948624704 compilation_cache.py:59] Initialized persistent compilation cache at /home/haha/.jax_compilation_cache
I0318 10:58:05.063319 140289948624704 xla_bridge.py:633] Unable to initialize backend 'cuda': Unable to load CUDA.
I0318 10:58:05.063727 140289948624704 xla_bridge.py:633] Unable to initialize backend 'rocm': NOT_FOUND: Could not find registered platform with name: "rocm". Available platform names are: CUDA
I0318 10:58:05.065797 140289948624704 xla_bridge.py:633] Unable to initialize backend 'tpu': INTERNAL: Failed to open libtpu.so: libtpu.so: cannot open shared object file: No such file or directory
W0318 10:58:05.065938 140289948624704 xla_bridge.py:697] CUDA backend failed to initialize: Unable to load CUDA. (Set TF_CPP_MIN_LOG_LEVEL=0 and rerun for more info.)
I0318 10:58:05.066064 140289948624704 finetune.py:63] 
        Octo Finetuning Script
        ======================
        Pretrained model: hf://rail-berkeley/octo-small
        Finetuning Dataset: bridge_dataset
        Data dir: ./tests/debug_dataset
        Task Modality: multimodal
        Finetuning Mode: full

        # Devices: 1
        Batch size: 256 (256 per device)
        # Steps: 50000
    
Fetching 8 files: 100%|█████████████████████████| 8/8 [00:00<00:00, 4329.05it/s]
I0318 10:58:13.746319 140289948624704 checkpointer.py:164] Restoring item from /home/haha/.cache/huggingface/hub/models--rail-berkeley--octo-small/snapshots/03d88976c54a58e10480d2043a8c762b35bc2611/270000/default.
I0318 10:58:14.635694 140289948624704 checkpointer.py:166] Finished restoring checkpoint from /home/haha/.cache/huggingface/hub/models--rail-berkeley--octo-small/snapshots/03d88976c54a58e10480d2043a8c762b35bc2611/270000/default.
I0318 10:58:16.732701 140289948624704 dataset_info.py:578] Load dataset info from tests/debug_dataset/bridge_dataset/1.0.0
I0318 10:58:16.845350 140289948624704 logging_logger.py:49] Constructing tf.data.Dataset bridge_dataset for split all, from tests/debug_dataset/bridge_dataset/1.0.0
I0318 10:58:17.743555 140289948624704 data_utils.py:110] Loading existing dataset statistics from tests/debug_dataset/bridge_dataset/1.0.0/dataset_statistics_38c366095a9a20c72a862f6a1a4a5ae4fe98ed4507d67cfa7314de76b872c6c4.json.
I0318 10:58:17.802858 140289948624704 logging_logger.py:49] Constructing tf.data.Dataset bridge_dataset for split train, from tests/debug_dataset/bridge_dataset/1.0.0
I0318 10:59:08.408026 140289948624704 train_utils.py:248] Freezing parameters that include the following keys: ['*hf_model*'].
I0318 10:59:08.411441 140289948624704 train_utils.py:284] Num trainable params: 27,042,060.
I0318 10:59:08.411522 140289948624704 train_utils.py:285] Num frozen params: 109,628,544.
I0318 10:59:08.411549 140289948624704 train_utils.py:286] To see a detailed list of frozen params, set logging level to DEBUG.
W0318 10:59:09.221674 140289948624704 finetune.py:261] save_dir not passed in, not saving checkpoints
I0318 10:59:09.225611 140289948624704 dataset_info.py:578] Load dataset info from tests/debug_dataset/bridge_dataset/1.0.0
I0318 10:59:09.294973 140289948624704 logging_logger.py:49] Constructing tf.data.Dataset bridge_dataset for split all, from tests/debug_dataset/bridge_dataset/1.0.0
I0318 10:59:09.553126 140289948624704 data_utils.py:110] Loading existing dataset statistics from tests/debug_dataset/bridge_dataset/1.0.0/dataset_statistics_38c366095a9a20c72a862f6a1a4a5ae4fe98ed4507d67cfa7314de76b872c6c4.json.
I0318 10:59:09.608441 140289948624704 logging_logger.py:49] Constructing tf.data.Dataset bridge_dataset for split val, from tests/debug_dataset/bridge_dataset/1.0.0
I0318 10:59:10.278485 140289948624704 dataset_info.py:578] Load dataset info from tests/debug_dataset/bridge_dataset/1.0.0
I0318 10:59:10.341788 140289948624704 logging_logger.py:49] Constructing tf.data.Dataset bridge_dataset for split all, from tests/debug_dataset/bridge_dataset/1.0.0
I0318 10:59:10.596746 140289948624704 data_utils.py:110] Loading existing dataset statistics from tests/debug_dataset/bridge_dataset/1.0.0/dataset_statistics_38c366095a9a20c72a862f6a1a4a5ae4fe98ed4507d67cfa7314de76b872c6c4.json.
I0318 10:59:10.650095 140289948624704 logging_logger.py:49] Constructing tf.data.Dataset bridge_dataset for split val, from tests/debug_dataset/bridge_dataset/1.0.0
  0%|                                                 | 0/50000 [00:00<?, ?it/s]2024-03-18 11:01:32.364956: E external/xla/xla/service/slow_operation_alarm.cc:65] 
********************************
[Compiling module jit_train_step] Very slow compile? If you want to file a bug, run with envvar XLA_FLAGS=--xla_dump_to=/tmp/foo and attach the results.
********************************
2024-03-18 11:03:52.031034: E external/xla/xla/service/slow_operation_alarm.cc:133] The operation took 4m19.665375883s

********************************
[Compiling module jit_train_step] Very slow compile? If you want to file a bug, run with envvar XLA_FLAGS=--xla_dump_to=/tmp/foo and attach the results.
********************************
  0%|                                   | 9/50000 [30:23<2447:58:38, 176.29s/it]
```

erikbr01, 2024-03-25:
I am having similar issues, trying to see what's going on. If I find a solution, I'll post it.

Xiaoyao-Li, 2025-01-08:
@erikbr01 Thanks, it helps for me!