octo-models/octo #72: What's the What is the model number of the UR5 robotic arm, camera and claw?

KangriX, 2024-04-09:
I want to test the performance of the octo through online simulation and am about to use RoboDK to simulate the UR5 robotic arm. Or is there another easier way for me to test the performance of the octo? Thanks for the help!

KangriX, 2024-04-09:
My purpose is to test the performance of octo as I am going to use model compression technique and I want to compare the performance of octo before and after compression. On-line simulation is one of the ways I've found so far, but I don't know the model of the robotic arm used in the experiment and the model of its tooling, its mounting location, etc. This is a very cumbersome task, so if there are other simpler ways to test the performance, please let me know immediately! Many thanks to everyone for their help!

kpertsch, 2024-04-09:
The best way to test Octo's performance is on a real robot system, eg the Bridge V2 system, which is fairly low-cost.
If you want to work in simulation, it is unlikely that the model will work out-of-the-box and you'll likely need to collect a few demonstrations in your sim environment and fine-tune the model, but then you could evaluate your model compression techniques on that fine-tuned version of Octo.

Leeviber, 2024-04-16:
Also have question about **_camera mounting location_**. If I want to test the model performance out-of-box on real UR5 robtic arm, do I need set up the observation space of UR5   exactly matches the observations in the dataset, such as camera position, arm mount position. 

For example the dataset `berkeley_autolab_ur5`, the main camera image as shown in beleow,  the camera is on the right side of the arm, if I want change the camera position to the top of the arm, is that OK? 
[image]

If not OK,  what kind of dataset for fine-tuning? Specifically, how should the images include the coverage of the arm? What is the ideal setup for this scenario?

Nimingez, 2024-05-16:
Have you ever figure out？ @Leeviber