octo-models/octo #89: UR wiping dataset

Nimingez, 2024-05-09:
Could you please provide the UR5 wiping dataset? I couldn't find it in Open X-Embodiment.
[image]

kpertsch, 2024-05-09:
It's the "Berkeley Autolab UR5" dataset

https://docs.google.com/spreadsheets/d/1rPBD77tk60AEIGZrGSODwyyzs5FgCU9Uz3h-3_t2A9g/edit?usp=sharing

[image]

Nimingez, 2024-05-09:
The data consists of 4 robot manipulation tasks: simple pick-and-place of a stuffed animal between containers, sweeping a cloth, stacking cups, and a more difficult pick-and-place of a bottle that requires precise grasp and 6DOF rotation 。did not contain wiping发自我的 iPhone

https://docs.google.com/spreadsheets/d/1rPBD77tk60AEIGZrGSODwyyzs5FgCU9Uz3h-3_t2A9g/edit?usp=sharing
image.png (view on web)

kpertsch, 2024-05-09:
Sweeping a cloth == wiping 

-- sorry I may not be understanding your question correctly?

Nimingez, 2024-05-10:
Thank you, I understand now

closed 2024-05-13 by Nimingez