octo-models/octo #94: How to load my local dataset

wuyukun-tong, 2024-05-18:
Hi,
I collected some data on our own robotics platform and processed it into the corresponding format using "bridge_rlds_builder", but didn't upload it to Google Netflix, and got an error when trying to load the background data using make_single_dataset.
# my code
dataset_kwargs = make_oxe_dataset_kwargs(
    
    "example_dataset",
    
    data_dir='/home/xx/tensorflow_datasets/example_dataset/1.0.0',
  
)
# error
raise ValueError(
  Cannot load {name} since only EEF pose delta action encoding is supported.

erikbr01, 2024-05-20:
You're going to have to write a spec for your dataset in oxe_dataset_configs.py, it's quite easy to extend it

wuyukun-tong, 2024-05-20:
thanks！ I have another question about "absolute_action_mask". In my data the action is robot speed how to set the value of "absolute_action_mask"?

kpertsch, 2024-05-20:
If all elements are velocity values (esp also for the gripper value) then you can set it to `False` everywhere

wuyukun-tong, 2024-05-21:
Thanks for your replay! The action space is velocity (vx,vy,vyaw) and control mode(0,1,2) .  May i set the (vx,vy,vyaw) to False and mode(0,1,2) to True ?

kpertsch, 2024-05-21:
jup that sounds reasonable

wuyukun-tong, 2024-05-21:
Thanks for your time

---- Replied Message ----
| From | Karl ***@***.***> |
| Date | 05/21/2024 10:29 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [octo-models/octo] How to load my local dataset (Issue #94) |

jup that sounds reasonable