octo-models/octo #113: about dataset_statistics

zhang-haojie, 2024-06-29:
When I run `03_eval_finetuned.py`, in lines 78, `octomodel.dataset_statistics` as a dictionary will index `['action']`, `model.dataset_statistics["action"]`. But when I check `dataset_statistics.json` in checkpoints, `action` is an attribute of a specific dataset. Is there any error in the code here?
[image]
[image]

dibyaghosh, 2024-06-29:
Great question! This is just a difference between the pre-trained model (which is trained on a mix of different datasets) and finetuned models (which is trained on a single dataset). 

For models trained on a mix of datasets (e.g. in pretraining), the `make_interleaved_dataset` returns the statistics for each dataset separately in a dict (so you must index  like model.dataset_statistics[dataset_name]['action']), but make_single_dataset (used in finetuning) directly returns the dataset statistics trained on a single dataset (so you access model.dataset_statistics['action'])

zhang-haojie, 2024-06-29:
Thank you for your timely answer to my question. It is already a very good code design!

closed 2024-06-29 by zhang-haojie