octo-models/octo #116: Action space clarification

abhyudit309, 2024-07-03:
Hi,

I wanted to confirm that the action space consists of the end-effector position and orientation displacement (along with the gripper state). Are these displacements/delta values in the world frame or the end-effector frame? Also, is there some way to know what the initial position of the robot arm is (from the datasets)?

Thanks!

peter-mitrano-bg, 2024-08-23:
Did you ever figure this out? I think knowing this is very important, and the original octo paper does not state the coordinate frame clearly.

peter-mitrano-bg, 2024-09-16:
I have looked at some of the datasets comprising OXE and I'm not convinced they are all the same... but yes I agree that most of them are deltas in the "base frame" (which will be different for every robot of course)