octo-models/octo #133: No GPU utilization with fine-tune example 02

zshanggu, 2024-10-05:
Hi, I'm using this command to run fine-tune example:
```
python examples/02_finetune_new_observation_action.py --pretrained_path=hf://rail-berkeley/octo-small-1.5 --data_dir="/home/DATA/example_sim_data/aloha_sim_dataset"  --save_dir="/home/octo/exp/
```
However, it seems the program doesn't consume any GPU memory-usage (2*4090), and is totally run on CPU.

(I can confirm that my GPU environment is good and it works well with the template fine-tune command
```
python scripts/finetune.py --config.pretrained_path=hf://rail-berkeley/octo-small-1.5 --config.save_dir="./exp/" --config.batch_size=128
```
)

Do you know what's the possible reason?

zshanggu, 2024-12-31:
Add below before running:

export XLA_PYTHON_CLIENT_PREALLOCATE=false
export XLA_PYTHON_CLIENT_MEM_FRACTION=.50
export XLA_PYTHON_CLIENT_ALLOCATOR=platform

zshanggu, 2025-01-18:
Plus: mismatched jax version vs cuda version

lostgreen, 2025-05-02:
@zshanggu Hi, how did you solve jax version mismatch?