octo-models/octo #159: AttributeError: 'Config' object has no attribute 'define_bool_state'

Luo-Zhongwei, 2025-03-29:
```python
>>> from octo.model.octo_model import OctoModel
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/autodl-tmp/octo/octo/model/octo_model.py", line 6, in <module>
    import flax
  File "/root/miniconda3/envs/svla/lib/python3.11/site-packages/flax/__init__.py", line 19, in <module>
    from .configurations import (
  File "/root/miniconda3/envs/svla/lib/python3.11/site-packages/flax/configurations.py", line 93, in <module>
    flax_filter_frames = define_bool_state(
                         ^^^^^^^^^^^^^^^^^^
  File "/root/miniconda3/envs/svla/lib/python3.11/site-packages/flax/configurations.py", line 42, in define_bool_state
    return jax_config.define_bool_state('flax_' + name, default, help)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Config' object has no attribute 'define_bool_state

```

wzc-handsome-boy, 2025-05-14:
the same question

Steven-Seven, 2025-07-28:
WZC已经收到你的邮件啦！！